tolstoyevsky/pieman #36: ubuntu-artful-arm64 build process returns Segmentation fault

dimapaloskin, 2018-01-24:
Hi. I trying to build `ubuntu-artful-arm64` image using next command

```sh
ALLOW_UNAUTHENTICATED=true BUILD_DIR=/build INCLUDES=vim,wireless-tools,wpasupplicant OS=ubuntu-artful-arm64  env ./pieman.sh
```

and that's what I see:
<details>
<summary>Full log</summary>

```sh
root@7c616d6efef2:~/pieman# ALLOW_UNAUTHENTICATED=true BUILD_DIR=/build INCLUDES=vim,wireless-tools,wpasupplicant OS=ubuntu-artful-arm64  env ./pieman.sh
+ ALLOW_UNAUTHENTICATED=true
+ BASE_DIR=
+ BUILD_DIR=/build
+ CREATE_ONLY_CHROOT=false
+ DEVICE=rpi-3-b
+ ENABLE_CUSTOM_DNS=
+ ENABLE_BASIC_YANDEX_DNS=false
+ ENABLE_FAMILY_YANDEX_DNS=false
+ ENABLE_GOOGLE_DNS=false
+ ENABLE_NONFREE=false
+ ENABLE_UNIVERSE=false
+ HOST_NAME=pieman_rpi-3-b
+ INCLUDES=vim,wireless-tools,wpasupplicant
+ OS=ubuntu-artful-arm64
+ CUSTOM_FILES=
+ PASSWORD=*****
++ uuidgen
+ PROJECT_NAME=e524728d-07a6-4caf-b2c4-960ede7e869a
+ PIEMAN_BIN=bin
++ which python3
+ PYTHON=/usr/bin/python3
+ TIME_ZONE=Etc/UTC
+ set +x
Info: using env DEBOOTSTRAP_DIR=/root/pieman/debootstrap ./debootstrap/debootstrap
gpg: keybox '/tmp/atomatically-generated-keyring.gpg' created
gpg: key 3B4FE6ACC0B21F32: public key "Ubuntu Archive Automatic Signing Key (2012) <ftpmaster@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: no ultimately trusted keys found
Info: running bootstrap/10-rootfs-partition.sh from bootstrap
Info: BASE_DIR is not specified or does not exist. Running debootstrap to create chroot environment.
Info: cannot run anything from devices/rpi-3-b/ubuntu-artful-arm64/pre-first-stage since it does not exist.
Info: FIRST STAGE
Info: env DEBOOTSTRAP_DIR=/root/pieman/debootstrap ./debootstrap/debootstrap --arch=arm64 --foreign --variant=minbase --keyring=/tmp/atomatically-generated-keyring.gpg artful /build/e524728d-07a6-4caf-b2c4-960ede7e869a/chroot http://ports.ubuntu.com/ubuntu-ports/
I: Retrieving InRelease
I: Checking Release signature
I: Valid Release signature (key id 790BC7277767219C42C86F933B4FE6ACC0B21F32)
I: Retrieving Packages
I: Validating Packages
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Found additional base dependencies: adduser gpgv libapt-pkg5.0 libffi6 libgmp10 libgnutls30 libhogweed4 libidn11 libnettle6 libp11-kit0 libstdc++6 libtasn1-6 ubuntu-keyring
I: Checking component main on http://ports.ubuntu.com/ubuntu-ports...
I: Retrieving adduser 3.113+nmu3ubuntu5
I: Validating adduser 3.113+nmu3ubuntu5
I: Retrieving apt 1.5
I: Validating apt 1.5
I: Retrieving base-files 9.6ubuntu102
I: Validating base-files 9.6ubuntu102
I: Retrieving base-passwd 3.5.43
I: Validating base-passwd 3.5.43
I: Retrieving bash 4.4-5ubuntu1
I: Validating bash 4.4-5ubuntu1
I: Retrieving bsdutils 1:2.30.1-0ubuntu4
I: Validating bsdutils 1:2.30.1-0ubuntu4
I: Retrieving coreutils 8.26-3ubuntu4
I: Validating coreutils 8.26-3ubuntu4
I: Retrieving dash 0.5.8-2.3ubuntu1
[... 241 lines cut ...]
+ umount -l /build/e524728d-07a6-4caf-b2c4-960ede7e869a/chroot/sys
+ /bin/true
+ umount /build/e524728d-07a6-4caf-b2c4-960ede7e869a/chroot/dev/pts
+ /bin/true
+ umount /build/e524728d-07a6-4caf-b2c4-960ede7e869a/chroot/tmp
+ /bin/true
+ check_if_variable_is_set LOOP_DEV
+ var_name=LOOP_DEV
+ '[' -z '' ']'
+ false
+ set +x
```

</details>

It's happens only with `--arch arm64`. I tested it only in docker environment.

```sh
root@7c616d6efef2:~/pieman# chroot /build/e524728d-07a6-4caf-b2c4-960ede7e869a/chroot id
Assertion failed: start < end (/home/buildozer/aports/main/qemu/src/qemu-2.8.1/translate-all.c: page_set_flags: 2098)
Segmentation fault
root@7c616d6efef2:~/pieman#
```

Any ideas?

eugulixes, 2018-01-24:
Hello Dima

Thank you for the report. Unfortunately, I can't reproduce the problem. I guess it's closely connected with the QEMU binaries you're using. So, I've got the following questions:
* Where did you get the QEMU binaries?
* What baseimage are using for the purpose?

eugulixes, 2018-06-30:
@dimapaloskin, the issue is related to Ubuntu 17.10. We've just upgraded 17.10 to 18.04, so I'm closing the issue since Ubuntu 17.10 is not supported by Pieman anymore. If the problem you described above is still occurs, feel free to open another issue.

closed 2018-06-30 by eugulixes