tolstoyevsky/pieman #85: Errors were encountered while processing:  initramfs-tools

mosesdeutschlaender, 2018-07-05:
Hi,
I'm trying to install ubuntu-bionic-arm64. debootstrap is (1.0.106).
But, I go to the following error:

> Setting up flash-kernel (3.90ubuntu3) ...
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76.)
debconf: falling back to frontend: Readline
Creating config file /etc/default/flash-kernel with new version
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Processing triggers for linux-image-4.15.0-1010-raspi2 (4.15.0-1010.11) ...
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-4.15.0-1010-raspi2
Warning: root device  does not exist
flash-kernel: deferring update (trigger activated)
/etc/kernel/postinst.d/zz-flash-kernel:
flash-kernel: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.130ubuntu3) ...
update-initramfs: Generating /boot/initrd.img-4.15.0-1010-raspi2
Warning: root device  does not exist
Unsupported platform.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
Info: build/dd1a2346-804f-42f9-b02a-3b00c0933f92/mount_point is not a mount point
Info: unmounting build/dd1a2346-804f-42f9-b02a-3b00c0933f92/chroot/proc
Info: unmounting build/dd1a2346-804f-42f9-b02a-3b00c0933f92/chroot/sys
Info: unmounting build/dd1a2346-804f-42f9-b02a-3b00c0933f92/chroot/dev/pts
Info: unmounting build/dd1a2346-804f-42f9-b02a-3b00c0933f92/chroot/dev
Info: unmounting build/dd1a2346-804f-42f9-b02a-3b00c0933f92/chroot/tmp
+++ rm -f /tmp/firstboot-dd1a2346-804f-42f9-b02a-3b00c0933f92.sh
+++ rm -f /tmp/atomatically-generated-keyring-for-dd1a2346-804f-42f9-b02a-3b00c0933f92.gpg
+++ check_if_variable_is_set LOOP_DEV
+++ var_name=LOOP_DEV
+++ '[' -z '' ']'
+++ false
+++ remove_temporary_dirs
+++ local target=build/dd1a2346-804f-42f9-b02a-3b00c0933f92
+++ rm -rf build/dd1a2346-804f-42f9-b02a-3b00c0933f92
+++ set +x

eugulixes, 2018-07-05:
@mosesdeutschlaender, thank you for opening the issue.
I can't reproduce the problem (even with debootstrap 1.0.106), but I think I know the root of it. Could you test the `bionic-fix-kernel-installation-in-chroot` branch?

I added the following patch which is applied before installation of the kernel.

```
diff --git a/functions b/functions
index b889c1a..6cc1943 100644
--- usr/share/flash-kernel/functions
+++ usr/share/flash-kernel/functions
@@ -710,6 +710,10 @@
 }
 
 main() {
+if systemd-detect-virt --chroot; then
+	echo "A chroot environment has been detected. Interrupt flash-kernel." >&2
+	exit 0
+fi
 force="no"
 if [ "x$1" = "x--force" ]; then
 	force="yes"
diff --git a/initramfs-tools/hooks/flash_kernel_set_root b/initramfs-tools/hooks/flash_kernel_set_root
index ac57d12..98d76ab 100755
--- usr/share/initramfs-tools/hooks/flash_kernel_set_root
+++ usr/share/initramfs-tools/hooks/flash_kernel_set_root
@@ -62,6 +62,11 @@ if systemd-detect-virt --quiet --container; then
 	exit 0
 fi
 
+# Do not run inside a chroot environment as well
+if systemd-detect-virt --chroot; then
+	exit 0
+fi
+
 # Record the root filesystem device for use during boot
 rootdev=$(egrep '^[^# 	]+[ 	]+/[ 	]' /etc/fstab | awk '{print $1}') || true
 
```

If it works fine, I'll merge it. Otherwise, we'll keep thinking how to fix the problem.

eugulixes, 2018-07-07:
@mosesdeutschlaender, I've just reproduced the problem and tested the patch. It works great and will be merged to `master` today.

commit 1c2d76fc (2018-07-07): Fix kernel installation in Ubuntu 18.04 based chroot

Fixes #85.
PR #89 (2018-07-07): Fix kernel installation in Ubuntu 18.04 based chroot
Fixes #85.
commit 70223cbf (2018-07-07): Fix kernel installation in Ubuntu 18.04 based chroot

Fixes #85.
closed 2018-07-07 by denismosolov