tolstoyevsky/pieman #279: Python support in  custom Image OS

Beebeejohn, 2020-11-16:
How to add python package/support to the image we are building using pieman or How to add python support to the existing OS we have created using pieman

eugulixes, 2020-11-16:
You can add Python to the image you're building via the `INCLUDES` param. For example, `sudo env INCLUDES=python3 ./pieman.sh`. Please have a look at the [documentation](https://github.com/tolstoyevsky/pieman#includes).
You can install Python 3 after building the image. Flash it to a memory card, boot it, login and run

```
sudo apt update
sudo apt install python3
```

Feel free to ask if you have any further questions.

Beebeejohn, 2020-11-17:
Thank you for the answer. Can u tell me how to add python daemon program in ./pieman.sh while building raspberry pi os image.  So that after flashing the OS, that daemon program should be running and it should display its output

eugulixes, 2020-11-17:
If it **is** a daemon, it **is not** capable of displaying anything. However, if you want to simply run a program on boot, add it to `files/etc/rc.local`.

Beebeejohn, 2020-11-18:
Thank you for the replies.  Will try it

eugulixes, 2020-11-18:
You're welcome. Closing the issue. Feel free to re-open it if you have further questions.

closed 2020-11-18 by eugulixes

Beebeejohn, 2020-11-18:
So for example if I want to add a1.py as a daemon, so the whole python code for a1.py should be added in files/etc/rc.local.  Was it like that?