tolstoyevsky/pieman #280: Installing python packages while building custom Raspberry pi OS

Beebeejohn, 2020-11-17:
How to install python packages like Adafruit.ads1x15.ads1115,busio,time,json,logging while building raspberry pi Os using piemann

eugulixes, 2020-11-17:
1. You have to add a `post-install` hook to `devices/rpi-2-b/raspbian-buster-armhf/post-install`. The hook might look like (I named it `run-pip3.sh`)
    ```bash
    #/bin/sh
    pip3 install adafruit-circuitpython-ads1x15 adafruit-circuitpython-bme280
    ```
2. Run `sudo env INCLUDES=python3-pip ./pieman.sh`

Beebeejohn, 2020-11-18:
Actually where this run-pip3.sh is there? Can you tell me the path.  I searched it .  But I could not get it.    And also this command where should I mention  **pip3 install adafruit-circuitpython-ads1x15 adafruit-circuitpython-bme280.**  Should I mention this line in pieman.sh script

eugulixes, 2020-11-18:
1. Go to `devices/rpi-2-b/raspbian-buster-armhf/`
2. Create a directory named `post-install` and go to the directory
3. Create a file named `run-pip3.sh` with the following contents
    ```bash
    #/bin/sh
    pip3 install adafruit-circuitpython-ads1x15 adafruit-circuitpython-bme280
    ```
4. Run `sudo env INCLUDES=python3-pip ./pieman.sh`

> And also this command where should I mention pip3 install adafruit-circuitpython-ads1x15 adafruit-circuitpython-bme280. Should I mention this line in pieman.sh script

No, you don't have to touch `pieman.sh`.

Beebeejohn, 2020-11-18:
Ok.  Thank you so much for the replies.  If I want to install multiple python packages like adafruit-blinka, adafruit-io, In the pip3.sh file, should I mention the contents as follows:
pip3 install adafruit-circuitpython-ads1x15 adafruit-circuitpython-bme280
pip3 install adafruit-blinka-bme280
pip3 install adafruit-io-bme280
Can you tell me

eugulixes, 2020-11-18:
yeah, you can either use multiple lines like that or put it in one line as `pip3 install adafruit-circuitpython-ads1x15 adafruit-circuitpython-bme280 Adafruit-Blinka adafruit-io`

Beebeejohn, 2020-11-18:
ya ok thank you

Beebeejohn, 2020-12-04:
I tried with the  above command as mentioned.  But adafruit-blinka and ads1115 libraries are not getting installed in python.  It is showing no module found error.  Can you tell me the exact command for installing supervisor,ssh and also  python packages like adafruit-blinka, etc., in a single command itself .  Because I need supervisor, ssh also along with python packages.  I tried with this command **sudo env INCLUDES=ssh,supervisor,python3-pip ./pieman.sh**.  But it is not working.  Can you tell me the command please? Whether do I need to use env also or not need

Beebeejohn, 2020-12-07:
Can anyone pls answer my above question?

eugulixes, 2020-12-07:
Fortunately or unfortunately I'm not able to read people's mind, so 1) explain what you did step by step, 2) attach some tracebacks or anything else what could help me or anyone else to help you. 

> Can you tell me the exact command for installing supervisor, ssh and also python packages like adafruit-blinka, etc., in a single command itself

There is no a single command because you have to install Python packages via pip3 and hooks as I mentioned above.

> Whether do I need to use env also or not needed?

It depends your task.

Beebeejohn, 2020-12-07:
The steps I followed are below:
1. I entered Wi-Fi SSID and password details in pieman.sh script and enabled wireless is equal to true then made DEVICE="rpi-zero-w"
2. I created new folder named **post-install** under devices/rpi-zero-w. (Because the raspberry pi, I am using for this experiement is **raspberry pi zero -w**) I created one file in that directory run-pip3.sh.  The following contents are there in that file 
**pip3 install adafruit-circuitpython-ads1x15 Adafruit-Blinka adafruit-io**
3. Then I tried installing python packages and supervisor in the terminal while building raspberry pi OS by the following command
 **sudo env INCLUDES=ssh,supervisor,python3-pip ./pieman.sh**
If I tried the above command, **python packages are not getting installed.**  Only supervisor is installed properly.  Now can you get my question what I am asking?

eugulixes, 2020-12-07:
Yeah, the problem is
1) You have to create `post-install` under `devices/rpi-b/raspbian-buster-armhf` (actually in `devices/rpi-zero-w/raspbian-buster-armhf` but it symlinks to `devices/rpi-b/raspbian-buster-armhf`).
2) You're building the image for RPi 3 but not for RPi Zero W. Try `sudo env DEVICE=rpi-zero-w INCLUDES=ssh,supervisor,python3-pip ./pieman.sh`

Beebeejohn, 2020-12-07:
Ok.  Will try like as u mentioned

Beebeejohn, 2020-12-08:
I tried with the above commands you mentioned.  But still python packages are not getting installed

eugulixes, 2020-12-08:
Attach the full build log. I need to see it to help you.

Beebeejohn, 2020-12-08:
The building steps followed are 
1. I entered Wi-Fi SSID and password details in pieman.sh script
2.I created new folder named post-install under devices/rpi-b. I created one file in that directory run-pip3.sh. The following contents are there in that file
pip3 install adafruit-circuitpython-ads1x15 Adafruit-Blinka adafruit-io
Then I tried installing python packages and supervisor in the terminal while building raspberry pi OS by the following command
**sudo env INCLUDES=python3-pip,ssh,supervisor,DEVICE=rpi-zero-w ENABLE_WIRELESS=true ./pieman.sh
**
Python packages like adafruit io, adafruit blink., etc., are not installed.  Showing like no module found error

eugulixes, 2020-12-08:
No, as I mentioned above you have to create the `post-install` directory under `devices/rpi-b/raspbian-buster-armhf` (**not** `devices/rpi-b` or anything else). Please, re-read the message carefully.

Beebeejohn, 2020-12-08:
I created that post-install directory under devices/rpi-b/raspbian-buster-armhf) only.  I havenot mentioned that full path here.
I reedited that.  Can you tell me now?

Beebeejohn, 2020-12-09:
Can you answer my question please

eugulixes, 2020-12-09:
I asked you in one of my messages above to attach the full build log. To do that, redirect stderr to a file. For example,

```
sudo env INCLUDES=python3-pip,ssh,supervisor,DEVICE=rpi-zero-w ENABLE_WIRELESS=true ./pieman.sh 2> buildlog
```

Then, attach the `buildlog` file. I will be able to study it and, probably, to help you with your problem.

Beebeejohn, 2020-12-09:
actually I did not get what is build log file.  Do u mean the  image which I built for built log file?

eugulixes, 2020-12-09:
I gave an example of how to get the build log file. Please, re-read my message carefully. In that example the the log file is called `buildlog`.

Beebeejohn, 2020-12-10:
Please find my build log file
[buildlog.zip](https://github.com/tolstoyevsky/pieman/files/5670570/buildlog.zip)
1. I tried with this command
**sudo INCLUDES=ssh,python3,python3-dev,python3-pip,python3-smbus,i2c-tools,nano,python-meld3,supervisor,raspi-config DEVICE=rpi-zero-w ENABLE_WIRELESS=true ENABLE_UNATTENDED_INSTALLATION=true ./pieman.sh 2> buildlog**
2. These contents are there in run-pip3.sh file which is created in devices/rpi-b/raspbian-buster-armhf/post-install
**sudo pip3 install board jsonlib-python3 sockets supervisor RPi.GPIO pyserial pyusb Adafruit_PureIO pyftdi Adafruit-PlatformDetect sysv-ipc rpi-ws281x  Adafruit-Blinka adafruit-circuitpython-busdevice adafruit-circuitpython-ads1x15**

Beebeejohn, 2020-12-11:
Can you answer my question?

eugulixes, 2020-12-11:
I will check it these weekends.