tolstoyevsky/pieman #281: Daemon program on boot while building raspberry pi custom OS

Beebeejohn, 2020-11-19:
So for example if I want to add a1.py as a daemon, so the whole python code content for a1.py should be added or pasted in files/etc/rc.local.  Was it like that? Can you tell me

_Originally posted by @Beebeejohn in https://github.com/tolstoyevsky/pieman/issues/279#issuecomment-729543109_

eugulixes, 2020-11-19:
You have to copy `a1.py` to the image somehow and then put `/path/to/a1.py` to `files/etc/rc.local`. However, Pieman doesn't support copying files to the images so far. It will be great if you implemented the feature in Pieman and sent us a PR. But still, you can copy `a1.py` to your image after building.

Beebeejohn, 2020-11-19:
Ok I will try it .  Thank you

Beebeejohn, 2020-11-23:
Actually If I mentioned the path of python code in files/etc/rc.loca/ for daemon process.  On boot, that daemon process is not working .  Can you tell me how to add daemon program before building raspberry pi OS

Beebeejohn, 2020-11-24:
Can someone tell me how to add daemon program while building raspberry pi os not after building raspberry pi OS since above method is not working

eugulixes, 2020-11-25:
I described above how it's _supposed_ to work. If it doesn't work as described it means there is a bug.