tolstoyevsky/pieman #283: SSH is not working from other PC's to Raspberry pi

Beebeejohn, 2020-11-23:
I have installed ssh by using sudo env INCLUDES=ssh pieman.sh.  But flashing the os, I am unable to do ssh from my PC to raspberry Pi whereas ssh is possible from raspberry pi board to other PC's.  Can anyone tell me how to solve this issue.  If I do ssh pi@ip address, it is showing like permission denied, although I have used same password used in pieman.sh script

eugulixes, 2020-11-23:
There is no any `pi` user in your image. Please, have a look at the `USER_NAME` param in the [documentation](https://github.com/tolstoyevsky/pieman#user_namecusdeb). The default value is `cusdeb`, so ssh using `ssh cusdeb@ip address`

Beebeejohn, 2020-11-24:
No actually I have changed username as pi in pieman.sh script and try giving ssh pi@ip address.  It is not working I meant.  Can you tell me how to fix this isssue

eugulixes, 2020-11-24:
You shouldn't edit `pieman.sh` manually. Use environment variables, instead. For example, `sudo USER_NAME=pi USER_PASSWORD=super-secret ./pieman.sh`

Beebeejohn, 2020-11-24:
Ok. Will try that. Similarly if I want to add SSID and password for wpa_suupplicant, should I mention as follows:, s**udo INCLUDES = WPA_SSID= **** WPA_PSK=pwd**,  or I can edit in pieman.sh.  Because previously for wifi, I edit pieman.sh manually, and raspberry pi gets connected to the Wi-fi details as I mentioned in pieman.sh script.  And also please tell me how to add daemon program for building raspberry pi Os

eugulixes, 2020-11-24:
> or I can edit in pieman.sh

You can but you shouldn't =) You shouldn't edit the source code of `pieman.sh` to simply pass some params to it. There is well documented interface based on environment variables. How I can help you when I don't even know what changes you made to the source code? I highly encourage you to modify the source code in case 1) it helps you understand how Pieman works and 2) you develop new features. 

> I edit pieman.sh manually, and raspberry pi gets connected to the Wi-fi details as I mentioned in pieman.sh script

You should use the `WPA_SSID` and `WPA_PSK` environment variables for it.

> And also please tell me how to add daemon program before building raspberry pi Os

I will answer in the corresponding issue (i.e. https://github.com/tolstoyevsky/pieman/issues/281).

Beebeejohn, 2020-11-24:
Is this command correct  for mentioning ssid and password **sudo INCLUDES = WPA_SSID= Rpi WPA_PSK=pwd, DEVICE=rpi-zero-w ./pieman.sh**.  I have not changed much in the source code.  In the source code, instead of rpi-3-b device, I mentioned as rpi-zero-w.  Similarly for WPA_SSid and PSK , I mentioned those credentials.  That's all.  Can you please tell me whether the above command is correct

eugulixes, 2020-11-24:
No, it doesn't seem to be correct. The reasons are
* There are spaces before and after `=`. It's totally wrong.
* `INCLUDES` seems to be empty in your example. Just remove it until it takes some value.

Beebeejohn, 2020-11-24:
Actually I did not get the 2nd point.  INCLUDES seem to be empty means?? Then how can I mention ssid details.  Now can you tell me whether this is correct.  I removed spaces **sudo INCLUDES=WPA_SSID= Rpi ****WPA_PSK=pwd,** DEVICE=rpi-zero-w ./pieman.sh**

eugulixes, 2020-11-24:
The `INCLUDES` param tells which packages must be installed to your operating system image. It's not connected with `WPA_SSID` and `WPA_PSK` at all. Please, have a look at the corresponding [section](https://github.com/tolstoyevsky/pieman#includes) of the documentation.

Beebeejohn, 2020-11-25:
Ya ok will look onto it .  Thank you for the reply

eugulixes, 2020-11-25:
You're welcome. Closing the issue. Feel free to re-open it if you have further questions.

closed 2020-11-25 by eugulixes