tolstoyevsky/pieman #284: SSH Issue

Beebeejohn, 2020-11-26:
As u mentioned in previous question,  I tried giving  different user name like this sudo INCLUDES=ssh USER_NAME=pi USER_PASSWORD=T@123 ./pieman.sh. But ssh is not working if I tried like ssh pi@ IP . Is this because of different username cusdeb and secret  as password mentioned in Pieman.sh  script.

Beebeejohn, 2020-11-26:
And also can u tell Me how to remove user name and password while doing login.  Because for removing username,  I made ENABLE_USER=false, and USER_NAME=" ", USER_PASSWORD=" ", in pieman.sh script and built image.  But still it is asking for username and password,  can u tell me how to remove username and password

eugulixes, 2020-11-26:
Currently it's not possible to log in automatically. In other words, Pieman provides you only with images which require login. However, you can google how to achieve yourself.

eugulixes, 2020-11-26:
> not working if I tried like ssh pi@ IP 

Where did you get `IP`?

Beebeejohn, 2020-11-26:
IP I will get from linux system. Like each system will be having different IP's like 10.75.1.2 something.  This IP is after flashing the OS in the system,  the raspberry pi is connected with PC.  IP is the system IP

Beebeejohn, 2020-12-01:
Can you please tell me answer for the first question why ssh is not working?

Beebeejohn, 2020-12-01:
The issue is we are able to **connect** the raspberry pi to other PC's** through ssh.  But if we want to do ssh from **other PC's to raspberry pi,** it is not happening.  Can anyone tell me how to fix this issue?

eugulixes, 2020-12-01:
Stop repeating the same question, please. I have already received it.
I need to check it and it will take some time. Please, be patient. Thank you.