tolstoyevsky/pieman #285: Installing supervisord

Beebeejohn, 2020-12-01:
How to install supervisord inside the image and to create supervisor configuration file  and also how to check the location where the libraries have been installed

eugulixes, 2020-12-01:
1. The package is called `supervisor`, so the correct command to prepare the image with supervisord installed is `sudo INCLUDES=supervisor ./pieman.sh`
2. Create a configuration file after building an image. It would be great if Pieman was making coffee or dealing with a supervisord configuration, but if wasn't so far=)
3. Check location of the libraries, executables, etc using https://packages.debian.org/buster/all/supervisor/filelist

eugulixes, 2020-12-07:
Closing the issue. Feel free to re-open it if you have further questions.

closed 2020-12-07 by eugulixes