tolstoyevsky/pieman #299: chroot: failed

flinxtm, 2021-03-28:
After call sudo ./pieman.sh

...
...
...

chroot: failed to run command ‘debootstrap/debootstrap’: Exec format error
Info: build/ba51878f-3a3a-45cd-a296-99e57dae42c1/mount_point is not a mount point
Info: build/ba51878f-3a3a-45cd-a296-99e57dae42c1/chroot/proc is not a mount point
Info: build/ba51878f-3a3a-45cd-a296-99e57dae42c1/chroot/sys is not a mount point
Info: build/ba51878f-3a3a-45cd-a296-99e57dae42c1/chroot/dev/pts is not a mount point
Info: build/ba51878f-3a3a-45cd-a296-99e57dae42c1/chroot/dev is not a mount point
Info: build/ba51878f-3a3a-45cd-a296-99e57dae42c1/chroot/tmp is not a mount point
+++ rm -f /tmp/firstboot-ba51878f-3a3a-45cd-a296-99e57dae42c1.sh
+++ rm -f /tmp/atomatically-generated-keyring-for-ba51878f-3a3a-45cd-a296-99e57dae42c1.gpg
+++ check_if_variable_is_set LOOP_DEV
+++ var_name=LOOP_DEV
+++ '[' -z '' ']'
+++ false
+++ remove_temporary_dirs
+++ local target=build/ba51878f-3a3a-45cd-a296-99e57dae42c1
+++ false
+++ rm -rf build/ba51878f-3a3a-45cd-a296-99e57dae42c1
+++ set +x

image is not building. Dir ./build is empty

eugulixes, 2021-03-29:
Thank you for the report.
I'm pretty sure the problem is that you don't have an emulator properly installed. Which operating system do you use?

flinxtm, 2021-03-30:
WSL2 Ubuntu 20.04 LTS
chroot and loop devices are working fine

eugulixes, 2021-03-30:
> WSL2 Ubuntu 20.04 LTS

Unfortunately, your case hasn't been tested and documented so far. I suggest you try to run a container based on the official Docker image as described [here](https://github.com/tolstoyevsky/pieman#docker).