tolstoyevsky/pieman #301: No space left on device

flinxtm, 2021-03-30:
sudo env DEVICE=rpi-3-b OS=raspbian-buster-armhf ENABLE_MENDER=true MENDER_DATA_SIZE=512 ./pieman.sh
...
...
...
cp: error writing 'build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22/mount_point/boot/zImage': No space left on device
Info: unmounting build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22/mount_point
Info: build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22/chroot/proc is not a mount point
Info: build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22/chroot/sys is not a mount point
Info: build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22/chroot/dev/pts is not a mount point
Info: build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22/chroot/dev is not a mount point
Info: build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22/chroot/tmp is not a mount point
++ rm -f /tmp/firstboot-ed44955b-c6fa-414e-b2d2-a50dcf48fc22.sh
++ rm -f /tmp/atomatically-generated-keyring-for-ed44955b-c6fa-414e-b2d2-a50dcf48fc22.gpg
++ check_if_variable_is_set LOOP_DEV
++ var_name=LOOP_DEV
++ '[' -z x ']'
++ true
++ losetup -d /dev/loop5
++ remove_temporary_dirs
++ local target=build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22
++ false
++ rm -rf build/ed44955b-c6fa-414e-b2d2-a50dcf48fc22
++ set +x

Same with Docker

eugulixes, 2021-03-31:
Thank you for reporting about the problem. I reproduced and got to the root of it. The problem is that the size of the image is calculated _before_ copying the Mender-related assets to the image, so there's no space for them. I'll try to fix the issue asap.

njskalski, 2022-08-20:
I got the same today :(